Xilinx/finn-examples #1: "mobilenetv1-w4a4.bit" Not Found

Wo-Ki, 2020-12-25:
Hi, When I run **2_imagenet_with_mobilenet_v1.ipynb**, line 2:
`accel = models.mobilenetv1_w4a4_imagenet()`
Error like:
`KeyError: 'finn_examples/bitfiles/Pynq-Z2/mobilenetv1-w4a4.bit'`
I found:
[image]
So, can you update this model? Thank you very much.

maltanar, 2020-12-29:
Which board are you trying to run this on? The MobileNet-v1 is currently only supported on the Alveo U250.

Wo-Ki, 2020-12-31:
> Which board are you trying to run this on? The MobileNet-v1 is currently only supported on the Alveo U250.

Oh，I see.My board is pynq-z2. I hope you can update it as fast as you can, I want to try this. Thank you.

maltanar, 2020-12-31:
Unfortunately the FPGA on the PYNQ-Z2 is too small to deploy MobileNet-v1 as a dataflow-style architecture so there are no plans to do this. We'll be releasing MobileNet-v1 for the ZCU104 but not for anything smaller.

Depending on your use case https://github.com/Xilinx/DPU-PYNQ may be of interest.

closed 2020-12-31 by maltanar