Xilinx/finn-examples #2: finn-examples not installable on Pynq-Z1 board

chhait07, 2021-02-12:
Hi everybody! I wanted to test some finn-examples from https://github.com/Xilinx/finn-examples on my Pynq-Z1-Board. I followed the instructions in section Quickstart.
1. Instruction `python3 -m pip install --upgrade pip setuptools` worked successfully.
2. The next instruction `pip3 install finn-examples` shows some errors, see the following screenshot.

[image]

Can you help me here?

maltanar, 2021-02-12:
Hi @chhait07 -- can you try this with root priviledges? Either `sudo pip3 install ...` or running `su` first prior to running `pip3 install`.

I've also moved this issue to `finn-examples` as this issue relates to that repo.

chhait07, 2021-02-18:
Hi @maltanar , I tested the command in both ways: `sudo pip3 install finn-examples` and `su` followed by `pip3 install finn-examples`. The output in both cases is the same as in my first comment.

Do you have any other ideas?

maltanar, 2021-02-18:
The core error in the screenshot from your first message seems to be "OSError: Root permissions required" which `sudo` really should resolve. If it doesn't there might be something wrong with your PYNQ installation -- make sure you are using the latest PYNQ image. If the problem persists I'd ask for help on the PYNQ forum: https://discuss.pynq.io/

chhait07, 2021-02-20:
Hi @maltanar, yes you are right, the output is different when executing the command with `sudo`. I'm sorry I've overlooked that. 
Here is the screenshot when trying with `sudo`:
[image]

maltanar, 2021-02-20:
Which PYNQ version are you using? If not the latest (v2.6.1) can you upgrade to the latest PYNQ version and try again?

chhait07, 2021-02-26:
Hi @maltanar. On my board PYNQ version 2.4 was running. I now updated it to version 2.6.0. I tried to update to 2.6.1 but that didn't work. I think due to [this article](https://discuss.pynq.io/t/install-pynq-v2-6-1-on-pynq-board/2352) version 2.6.0 should be sufficient. If not, can you tell me how to update to V2.6.1?

I again tried to install finn-examples with the following result:
[image]

There seems to be missing a package or directory. What can I do to solve the problem?

maltanar, 2021-03-02:
Does `python3 -m pip install --upgrade pip setuptools` help?

chhait07, 2021-03-05:
Hi @maltanar , I tried `sudo python3 -m pip install --upgrade pip setuptools` but this exited with an error. When I omit the `sudo` I can successfully upgrade pip setuptools. See screenshot:
[image]

I then tried to install finn-examples again, with the same output as in my last comment:
[image]

maltanar, 2021-03-09:
It looks like you are having connection issues while upgrading setuptools above (`Temporary failure in name resolution`). Try again with `sudo` on both commands. If the problem persists I'd ask for help on the PYNQ forum: https://discuss.pynq.io/

chhait07, 2021-03-14:
Hi @maltanar , today I tried again installing the setuptools with `sudo python3 -m pip install --upgrade pip setuptools` . This time it was successfully installed without errors (I have no idea why it is successful this time). My board is now updated from setuptools 39.0.1 to 54.1.1
Now I can successfully install finn-examples! Thank you @maltanar for your help!

Unfortunately I have now another problem on retrieving the example Jupyter notebooks. Thus I opened a [new issue](https://github.com/Xilinx/finn-examples/issues/6) and I will close this issue.

closed 2021-03-14 by chhait07