Xilinx/finn-examples #6: Can not retrieve the example Jupyter notebooks

chhait07, 2021-03-14:
Hi everybody,
I'm currently trying to setup finn-examples on my Pynq-Z1 board. I successfully executed `python3 -m pip install --upgrade pip setuptools` and `pip3 install finn-examples`. On the next step I try `pynq get-notebooks --from-package finn-examples -p .` command and receive an error. Executing the same command with sudo it says that no notebooks are available. See the following screenshot. 

[image]

Can somebody help me here?

universe-12, 2021-04-09:
I had the same problem as you posted it, I had an error installing finn-examples because of the setuptools
so I've to upgrade setuptools with this command,

`sudo pip install -U setuptools`
and
`cd $(finn-examples)`
`sudo pip3 install .`  to install finn-examples
And then I try 'sudo pynq get-notebooks --from-package finn-examples -p .'  and it works well

chhait07, 2021-04-10:
Hi, thanks for your reply.
I tried it your way and it worked for me!
Insted of `cd $(finn-examples)` - which didn't work for me - I searched for the install path of finn-examples and executed commands `sudo pip3 install .` there. Then I can do `sudo pynq get-notebooks --from-package finn-examples -p .` and it works fine.

Thank you very much!

closed 2021-04-10 by chhait07

royliuyu, 2021-04-16:
@universe-12  It's also solved my the same question. Realy thanks.