Xilinx/finn-examples #13: Rebuild resnet50 failed in step_resnet50_set_fifo_depths [9/13]

qpchen, 2021-06-21:
When I rebuild resnet50 following the readme. After I run "./run-docker.sh build_custom ~/master/develop/finn-examples/build/resnet50", the script throw error and stop into pdb.
The Error information is as follows:

```
Running step: step_resnet50_set_fifo_depths [9/13]
Traceback (most recent call last):
  File "/workspace/finn/src/finn/builder/build_dataflow.py", line 128, in build_dataflow_cfg
    model = transform_step(model, cfg)
  File "/workspace/master/develop/finn-examples/build/resnet50/custom_steps.py", line 274, in step_resnet50_set_fifo_depths
    model = model.transform(
  File "/home/jumpchan/.local/lib/python3.8/site-packages/finn/core/modelwrapper.py", line 139, in transform
    (transformed_model, model_was_changed) = transformation.apply(
  File "/workspace/finn/src/finn/transformation/fpgadataflow/set_fifo_depths.py", line 322, in apply
    current_state = sim.internals["finn_design_i"][key]["inst"][
  File "/workspace/pyverilator/pyverilator/pyverilator.py", line 170, in __getitem__
    raise ValueError("'Collection' object has no item '%s'" % name)
ValueError: 'Collection' object has no item 'StreamingFIFO_1'
> /workspace/pyverilator/pyverilator/pyverilator.py(170)__getitem__()
-> raise ValueError("'Collection' object has no item '%s'" % name)
(Pdb)
```
Looks like the "StreamingFIFO_1" is not found. But I can not locate who wants "StreamingFIFO_1", and where the "StreamingFIFO_1" item should be? Could you help me with that?

My environment is:

> Ubuntu 18.04
> Vitis 2020.1
> XRT_DEB_VERSION=xrt_202010.2.6.655_18.04-amd64-xrt
> Card is U250

Looking forward to your reply! Thank you for your time!

qpchen, 2021-06-28:
There are some replies at gitter community room as follows, but not finished for now.
[click to look at the same issue at Gitter finn/community room](https://gitter.im/xilinx-finn/community?at=60d44ca68c12474d8ccd23cf)

laochonlam, 2021-08-13:
Same situation here. Looking forward to the patch!

lloo099, 2021-11-02:
same problem as me.

Building dataflow accelerator from model.onnx
Intermediate outputs will be generated in /tmp/finn_dev_enai
Final outputs will be generated in output_tfc_w1a1_fpga
Build log is at output_tfc_w1a1_fpga/build_dataflow.log
Running step: step_tidy_up [1/16]
Running step: step_streamline [2/16]
Running step: step_convert_to_hls [3/16]
Running step: step_create_dataflow_partition [4/16]
Running step: step_target_fps_parallelization [5/16]
Running step: step_apply_folding_config [6/16]
Running step: step_generate_estimate_reports [7/16]
Running step: step_hls_codegen [8/16]
Running step: step_hls_ipgen [9/16]
Running step: step_set_fifo_depths [10/16]
ERROR: [BD 5-683] VLNV <xilinx.com:user:memstream:1.0> is not supported for the current part.
Traceback (most recent call last):
  File "/workspace/finn/src/finn/builder/build_dataflow.py", line 128, in build_dataflow_cfg
    model = transform_step(model, cfg)
  File "/workspace/finn/src/finn/builder/build_dataflow_steps.py", line 331, in step_set_fifo_depths
    model = model.transform(
  File "/workspace/finn-base/src/finn/core/modelwrapper.py", line 139, in transform
    (transformed_model, model_was_changed) = transformation.apply(
  File "/workspace/finn/src/finn/transformation/fpgadataflow/set_fifo_depths.py", line 300, in apply
    sim = pyverilate_stitched_ip(model)
  File "/workspace/finn-base/src/finn/util/pyverilator.py", line 155, in pyverilate_stitched_ip
    with open(vivado_stitch_proj_dir + "/all_verilog_srcs.txt", "r") as f:
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/finn_dev_enai/vivado_stitch_proj_r95_h97s/all_verilog_srcs.txt'
> /workspace/finn-base/src/finn/util/pyverilator.py(155)pyverilate_stitched_ip()
-> with open(vivado_stitch_proj_dir + "/all_verilog_srcs.txt", "r") as f:
(Pdb)

fpjentzsch, 2021-11-02:
Hi @lloo099, your issue is not related to the original issue, but seems to be the same issue as https://github.com/Xilinx/finn-examples/issues/21.

The original issue has already been fixed in https://github.com/Xilinx/finn-base/pull/33, so we can close this one.

closed 2021-11-03 by fpjentzsch