Xilinx/finn-examples #17: Where did the ResNet W1A2 ONNX model files come from?

laochonlam, 2021-08-01:
Hi, I am trying to make some performance comparisons between the GPU solution and FINN. 

May I know where this w1a2 ResNet50 ONNX model comes from? I couldn't find it in Brevitas examples. Thanks!

https://github.com/Xilinx/finn-examples/tree/main/build/resnet50

maltanar, 2021-08-02:
Hello @laochonlam, the ResNet-50 model was trained by my colleague @nickfraser and I believe the clean-up is still ongoing for releasing this as part of the Brevitas examples. In the meantime you can read more about how this model was trained in Section III of  https://arxiv.org/pdf/2011.07317.pdf

laochonlam, 2021-08-05:
Thanks @maltanar. Looking forward to the release!

closed 2021-08-05 by laochonlam