Xilinx/finn-examples #21: Rebuild tfc model failed on stitched IP

lloo099, 2021-11-02:
Hi, I encounter one error shows below
[image]

fpjentzsch, 2021-11-02:
Hi,
which device/part are you targeting with this build?

lloo099, 2021-11-02:
> Hi, which device/part are you targeting with this build?

fpga_part="xqrku060-cna1509-1M-m",
device: PYNQ-Z1

lloo099, 2021-11-02:
Device: ZCU102

And I also run the mobilenetv1 example, error happened on step11:

Building dataflow accelerator from models/mobilenetv1-w4a4_pre_post_tidy.onnx
Intermediate outputs will be generated in /tmp/finn_dev_enai
Final outputs will be generated in output_mobilenetv1-w4a4_ZCU102
Build log is at output_mobilenetv1-w4a4_ZCU102/build_dataflow.log
Running step: step_mobilenet_streamline [1/13]
Running step: step_mobilenet_lower_convs [2/13]
Running step: step_mobilenet_convert_to_hls_layers_separate_th [3/13]
Running step: step_create_dataflow_partition [4/13]
Running step: step_apply_folding_config [5/13]
Running step: step_generate_estimate_reports [6/13]
Running step: step_hls_codegen [7/13]
Running step: step_hls_ipgen [8/13]

Running step: step_set_fifo_depths [9/13]
Running step: step_create_stitched_ip [10/13]
Running step: step_synthesize_bitfile [11/13]
ERROR: [Common 17-69] Command failed: Run 'synth_1' has not been launched. Unable to open

lloo099, 2021-11-02:
> Hi, which device/part are you targeting with this build?

This problem has solved. As I modify fpga-part command. But mobilenetv1 still has 'synth_1' problems

lloo099, 2021-11-02:
> Device: ZCU102
>
> And I also run the mobilenetv1 example, error happened on step11:
>
> Building dataflow accelerator from models/mobilenetv1-w4a4_pre_post_tidy.onnx Intermediate outputs will be [...]

Running step: step_synthesize_bitfile [11/13]
ERROR: [Common 17-69] Command failed: Run 'synth_1' has not been launched. Unable to open

ERROR: [Common 17-69] Command failed: Run 'synth_1' has not been launched. Unable to open
ERROR: [Common 17-69] Command failed: Run 'synth_1' has not been launched. Unable to open
ERROR: [BD 5-390] IP definition not found for VLNV: xilinx_finn:finn:StreamingDataflowPartition_0:1.0 
Traceback (most recent call last):
  File "/workspace/finn/src/finn/builder/build_dataflow.py", line 128, in build_dataflow_cfg
    model = transform_step(model, cfg)
  File "/workspace/finn/src/finn/builder/build_dataflow_steps.py", line 491, in step_synthesize_bitfile
    model = model.transform(
  File "/workspace/finn-base/src/finn/core/modelwrapper.py", line 139, in transform
    (transformed_model, model_was_changed) = transformation.apply(
  File "/workspace/finn/src/finn/transformation/fpgadataflow/make_zynq_proj.py", line 343, in apply
    model = model.transform(
  File "/workspace/finn-base/src/finn/core/modelwrapper.py", line 139, in transform
    (transformed_model, model_was_changed) = transformation.apply(
  File "/workspace/finn/src/finn/transformation/fpgadataflow/make_zynq_proj.py", line 268, in apply
    raise Exception(
Exception: Synthesis failed, no bitfile found. Check logs under /tmp/finn_dev_enai/vivado_zynq_proj_6x3erjya
> /workspace/finn/src/finn/transformation/fpgadataflow/make_zynq_proj.py(268)apply()
-> raise Exception(

closed 2021-11-25 by lloo099