Xilinx/finn-examples #27: finn build problem : Running step: step_synthesize_bitfile

lloo099, 2021-11-25:
Hi @maltanar and all, I encounter some errors about synth_1 problems. And u may see below information and report_ip_status, it seems some ip cannot be found. But I follow your steps to generate finn ips. How to deal with it? thanks,

Platform: U200
model: Mobilenetv1
Ubuntu:18.04
Vitis:2020.1

```
Running step: step_synthesize_bitfile [11/13]
ERROR: [Common 17-69] Command failed: Run 'synth_1' has not been launched. Unable to open
ERROR: [Common 17-69] Command failed: The provided IP directory doesn't exist: '/tmp/finn_dev_enai/vivado_stitch_proj_82a0t6qf/ip'
Traceback
![hls_mobilenetv1_error1](https://user-images.githubusercontent.com/29093207/143428638-ade83f41-785d-4687-9ab2-992c81ba2992.jpg)
 (most recent call last):
  File "/workspace/finn/src/finn/builder/build_dataflow.py", line 166, in build_dataflow_cfg
    model = transform_step(model, cfg)
  File "/workspace/finn/src/finn/builder/build_dataflow_steps.py", line 643, in step_synthesize_bitfile
    model = model.transform(
  File "/workspace/finn-base/src/finn/core/modelwrapper.py", line 141, in transform
    (transformed_model, model_was_changed) = transformation.apply(
  File "/workspace/finn/src/finn/transformation/fpgadataflow/vitis_build.py", line 431, in apply
    kernel_mode
![hls_mobilenetv1_error1](https://user-images.githubusercontent.com/29093207/143428704-1771094d-7a8a-4627-9d98-b568f002d509.jpg)
l = kernel_model.transform(
  File "/workspace/finn-base/src/finn/core/modelwrapper.py", line 141, in transform
    (transformed_model, model_was_changed) = transformation.apply(
  File "/workspace/finn/src/finn/transformation/fpgadataflow/vitis_build.py", line 163, in apply
    assert os.path.isfile(xo_path), (
AssertionError: Vitis .xo file not created, check logs under /tmp/finn_dev_enai/vivado_stitch_proj_82a0t6qf
> /workspace/finn/src/finn/transformation/fpgadataflow/vitis_build.py(163)apply()
-> assert os.path.isfile(xo_path), (

```

```
Copyright 1986-2020 Xilinx, Inc. All Rights Reserved.
------------------------------------------------------------------------------------
| Tool Version : Vivado v.2020.1 (lin64) Build 2902540 Wed May 27 19:54:35 MDT 2020
| Date         : Thu Nov 25 18:49:05 2021
| Host         : enai-System-Product-Name running 64-bit Ubuntu 18.04.5 LTS
| Command      : report_ip_status
------------------------------------------------------------------------------------

IP Status Summary

1. Project IP Status
--------------------
Your project uses 46 IP. Some of these IP may have undergone changes in this release of the software. Please review the recommended actions.

More information on the Xilinx versioning policy is available at www.xilinx.com.

Project IP Instances
+--------------------------------------------------------------------+-------------------------+------------------------------+-----------+--------------------+---------+-----------------------+------------+----------------------+
| Instance Name                                                      | Status                  | Recommendation               | Change    | IP Name            | IP      | New Version           | New        | Original Part        |
|                                                                    |                         |                              | Log       |                    | Version |                       | License    |                      |
+--------------------------------------------------------------------+-------------------------+------------------------------+-----------+--------------------+---------+-----------------------+------------+----------------------+
| StreamingDataflowPartition_5_ChannelwiseOp_Batch_0_0               | Up-to-date              | No changes required          | Change    | Channelwiseop_batc | 1.0     | 1.0 (Rev. 2111250721) | Included   | xcu200-fsgd2104-2-e  |
|                                                                    |                         |                              | Log not   |                    | (Rev.   |                       |            |                      |
|                                                                    |                         |                              | available |                    | 2111250 |                       |            |                      |
+--------------------------------------------------------------------+-------------------------+------------------------------+-----------+--------------------+---------+-----------------------+------------+----------------------+
| StreamingDataflowPartition_5_ConvolutionInputGenerator_0_0         | Up-to-date              | No changes required          | Change    | Convolutioninputge | 1.0     | 1.0 (Rev. 2111250733) | Included   | xcu200-fsgd2104-2-e  |
|                                                                    |                         |                              | Log not   |                    | (Rev.   |                       |            |                      |
|                                                                    |                         |                              | available |                    | 2111250 |                       |            |                      |
+--------------------------------------------------------------------+-------------------------+------------------------------+-----------+--------------------+---------+-----------------------+------------+----------------------+
| StreamingDataflowPartition_5_ConvolutionInputGenerator_1_0         | Up-to-date              | No changes required          | Change    | Convolutioninputge | 1.0     | 1.0 (Rev. 2111250729) | Included   | xcu200-fsgd2104-2-e  |
|                                                                    |                         |                              | Log not   |                    | (Rev.   |                       |            |                      |
|                                                                    |                         |                              | available |                    | 2111250 |                       |            |                      |
+--------------------------------------------------------------------+-------------------------+------------------------------+-----------+--------------------+---------+-----------------------+------------+----------------------+
| StreamingDataflowPartition_5_ConvolutionInputGenerator_2_0         | Up-to-date              | No changes required          | Change    | Convolutioninputge | 1.0     | 1.0 (Rev. 2111250724) | Included   | xcu200-fsgd2104-2-e  |
|                                                                    |                         |                              | Log not   |                    | (Rev.   |                       |            |                      |
[... 154 lines cut ...]
| StreamingDataflowPartition_5_fifo_3                                | Up-to-date              | No changes required          |  *(4)     | AXI4-Stream Data   | 2.0     | 2.0 (Rev. 3)          | Included   | xcu200-fsgd2104-2-e  |
|                                                                    |                         |                              |           | FIFO               | (Rev.   |                       |            |                      |
|                                                                    |                         |                              |           |                    | 3)      |                       |            |                      |
+--------------------------------------------------------------------+-------------------------+------------------------------+-----------+--------------------+---------+-----------------------+------------+----------------------+
| StreamingDataflowPartition_5_fifo_4                                | Up-to-date              | No changes required          |  *(5)     | AXI4-Stream Data   | 2.0     | 2.0 (Rev. 3)          | Included   | xcu200-fsgd2104-2-e  |
|                                                                    |                         |                              |           | FIFO               | (Rev.   |                       |            |                      |
|                                                                    |                         |                              |           |                    | 3)      |                       |            |                   [... cut ...]
*(3) /tools/Xilinx/Vivado/2020.1/data/ip/xilinx/axis_data_fifo_v2_0/doc/axis_data_fifo_v2_0_changelog.txt
*(4) /tools/Xilinx/Vivado/2020.1/data/ip/xilinx/axis_data_fifo_v2_0/doc/axis_data_fifo_v2_0_changelog.txt
*(5) /tools/Xilinx/Vivado/2020.1/data/ip/xilinx/axis_data_fifo_v2_0/doc/axis_data_fifo_v2_0_changelog.txt
*(6) /tools/Xilinx/Vivado/2020.1/data/ip/xilinx/axis_data_fifo_v2_0/doc/axis_data_fifo_v2_0_changelog.txt
*(7) /tools/Xilinx/Vivado/2020.1/data/ip/xilinx/axis_data_fifo_v2_0/doc/axis_data_fifo_v2_0_changelog.txt

```

lloo099, 2021-11-25:
[image]

lloo099, 2021-12-06:
hi @maltanar , could u help me with this? thanks

maltanar, 2021-12-06:
Hi @lloo099 , can you describe the steps you are following to get this error? 

If you are running `report_ip_status` outside Docker, the errors about not finding the weight streamer IP can be a false alarm because the path for that IP is normally specified to be inside the Docker container (`/workspace/finn/finn-rtllib/memstream`). When opening the same project outside of Docker for synthesis or debugging, you'll need to add the host-side path (something like `/path/to/finn/finn-rtllib/memstream`) to the list of IP repos.

lloo099, 2021-12-07:
> Hi @lloo099 , can you describe the steps you are following to get this error?
>
> If you are running `report_ip_status` outside Docker, the errors about not finding the weight streamer IP can be a [...]

Hi, I following this
```
export VIVADO_PATH=/tools/Xilinx/Vivado/2020.1
export FE_PATH=/home/enai/Desktop/project/finn-examples-main/finn_examples
export PYNQ_IP=192.168.1.7
export VITIS_PATH=/tools/Xilinx/Vitis
source /opt/xilinx/xrt/setup.sh
export PLATFORM_REPO_PATHS=/opt/xilinx/platforms
export FINN_XILINX_PATH=/tools/Xilinx
export FINN_XILINX_VERSION=2020.1
./run-docker.sh build_custom /home/enai/Desktop/project/finn/build/mobilenetv1
```

lloo099, 2021-12-07:
> Hi, I following this
>
> ```
> export VIVADO_PATH=/tools/Xilinx/Vivado/2020.1
> export FE_PATH=/home/enai/Desktop/project/finn-examples-main/finn_examples
> export PYNQ_IP=192.168.1.7
> export [...]

btw, I cannot find this fold "vivado_stitch_proj_82a0t6qf". Do u know how to generate it?

lloo099, 2021-12-07:
Does it failed due to insufficient memory
```
[10:08:13] Starting bitstream generation..
[10:13:22] Run vpl: Step impl: Failed
[10:13:24] Run vpl: FINISHED. Run Status: impl ERROR
ERROR: [VPL 60-773] In '/tmp/finn_dev_enai/vitis_link_proj_3oie5ezy/_x/link/vivado/vpl/vivado.log', caught Tcl error:  problem implementing dynamic region, impl_1: route_design ERROR, please look at the run log file '/tmp/finn_dev_enai/vitis_link_proj_3oie5ezy/_x/link/vivado/vpl/prj/prj.runs/impl_1/runme.log' for more information
WARNING: [VPL 60-732] Link warning: No monitor points found for BD automation.
ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, impl_1: route_design ERROR, please look at the run log file '/tmp/finn_dev_enai/vitis_link_proj_3oie5ezy/_x/link/vivado/vpl/prj/prj.runs/impl_1/runme.log' for more information
ERROR: [VPL 60-1328] Vpl run 'vpl' failed
ERROR: [VPL 60-806] Failed to finish platform linker
INFO: [v++ 60-1442] [10:13:24] Run run_link: Step vpl: Failed
Time (s): cpu = 00:02:47 ; elapsed = 04:32:39 . Memory (MB): peak = 1336.637 ; gain = 0.000 ; free physical = 23993 ; free virtual = 24457
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
INFO: [v++ 60-1653] Closing dispatch client.

```

maltanar, 2021-12-07:
Looking at the runme.log you shared, it ends with:

`/tools/Xilinx/Vivado/2020.1/bin/loader: line 286:  6912 Killed                  "$RDI_PROG" "$@"`

which is probably because it's running out of memory. Vitis/Alveo builds need a lot of memory (at least 64GB, or more if you are running other things at the same time / using the `NUM_DEFAULT_WORKERS` env.var. for build parallelism) https://finn.readthedocs.io/en/latest/getting_started.html#system-requirements

> btw, I cannot find this fold "vivado_stitch_proj_82a0t6qf". Do u know how to generate it?

`vivado_stitch_proj_82a0t6qf` should be under `/tmp/finn_dev_enai` -- you can also specify the `FINN_HOST_BUILD_DIR` environment variable to place those generated files somewhere else before running the script.

lloo099, 2021-12-07:
> Looking at the runme.log you shared, it ends with:
>
> `/tools/Xilinx/Vivado/2020.1/bin/loader: line 286: 6912 Killed "$RDI_PROG" "$@"`
>
> which is probably because it's running out of memory. [...]

Hi, thanks. It seems ur right. My RAM is only 32GB. So does it have a new way to build it on 32GB? Or I must to upgrade my RAM to least 64GB.

closed 2021-12-08 by lloo099