Xilinx/finn-examples #28: 2_imagenet_with_cnns.ipynb stuck

laochonlam, 2021-12-12:
Hi FINN community,

When I ran `2_imagenet_with_cnns.ipynb` I found out that it stucked when I set batch size other than 1. The program will be hang on `obuf_normal = accel.execute(ibuf_normal)`. 

[image]

Do you have any clue for this?

Thanks,
Lam

mmrahorovic, 2023-02-14:
Hi @laochonlam, thank you for your post!
Unfortunately, I have not encountered this issue before. However, I would advise to give the latest [FINN-examples release](https://github.com/Xilinx/finn-examples/tree/v0.0.6) a try. Please let me know if you still observe the same problem.

auphelia, 2024-08-16:
Closing this issue for now, please feel free to re-open if your problem isn't solved.

closed 2024-08-16 by auphelia