Xilinx/finn-examples #30: ResNet-50 model and notebook release

eneneli, 2022-04-28:
Hi, I am wondering about the release mentioned in issue #17. 

I have run into some problems implementing ResNet-18 with FINN and it would be very helpful to see the PyTorch/Brevitas model as well as the notebook with the transformations and steps applied to arrive at the ResNet-50 bitstream.

Is there any chance for this code to be added to the repository anytime soon?

mmrahorovic, 2023-02-14:
Hi @eneneli! 
The Brevitas code for ResNet-50 is currently not released, but we are working on it and we will let you know in the upcoming months when it is ready!
For the transformation steps, you can find all details [here](https://github.com/Xilinx/finn-examples/tree/main/build/resnet50) in the `build.py` and `custom_steps.py` files.
Please let me know if you have any other question!

0BAB1, 2025-01-08:
Hi everyone,

Any updates on this ? Is it possible to implement different kind of networks that uses residual nodes ?

penrosed, 2025-01-14:
It's definitely possible use FINN with residual networks, but it'll take some work to get there.

There is a branch of this repository, [feature/rn-18](https://github.com/Xilinx/finn-examples/tree/feature/rn-18), which contains a [build script for ResNet18](https://github.com/Xilinx/finn-examples/blob/feature/rn-18/build/resnet18/build.py), and the [Brevitas-trained model](https://github.com/Xilinx/finn-examples/blob/feature/rn-18/build/resnet18/models/resnet18_4w4a.onnx) it takes as an input.

This branch isn't yet ready for release, as it has not yet been extensively tested, and is unfinished in places. It's likely you'll encounter some problems trying to use it. However, you could use this branch, in conjunction with the already existing [ResNet18 Brevitas example](https://github.com/Xilinx/brevitas/blob/dev/src/brevitas_examples/bnn_pynq/models/resnet.py) (which was used to produce model mentioned above), and get a decent idea of how to use FINN end-to-end with a residual network.

0BAB1, 2025-01-15:
Thanks for the input !

Davorin51, 2025-01-30:
Hello everyone,

I am trying to deploy a ResNet18 model on ZCU104 using FINN, but I am facing some issues during the build process. Specifically:

- I have adapted the build script for ZCU104 with folding config.
- The synthesis process fails at the **CreateStitchedIP** step with an "IP definition not found error (e.g., `AddStreams_hls_6` missing)".
- I noticed that `split_large_fifos=True` might be creating additional nodes that are not getting generated correctly.

Has anyone successfully deployed ResNet18 on ZCU104?