Xilinx/finn-examples #34: Installing finn-examples fails on Pynq Z2 (version 2.7)

developandplay, 2022-06-24:
After managing to rebuild the BNN images for the Pynq Z2 boards I'm having issues with deploying them.
Unfortunatelfy the `pip3 install git+https://github.com/Xilinx/finn-examples.git@dev -v` command fails with the following error:
```
arm-linux-gnueabihf-gcc: fatal error: Killed signal terminated program as
    compilation terminated.
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
    ----------------------------------------
    ERROR: Failed building wheel for pandas
  Successfully built pynq cffi numpy
  Failed to build pandas
  ERROR: Could not build wheels for pandas which use PEP 517 and cannot be installed directly
```

Are there any steps missing in the quickstart section of the README?

Complete log can be found here: https://gist.github.com/developandplay/128bb4974966d83b2a1217d26008ff60

developandplay, 2022-06-25:
So the real issue is that the installation of finn-examples ignores all previous installed dependencies and reinstalls them.
Of course when I install the dependencies manually I can download the wheels from [here](https://wheel-index.linuxserver.io/ubuntu/) and install them quickly.
However the install script from finn-examples discards all of this and tries to reinstall all packages from source which the PYNQ Z2 is simply not powerful enough.

@maltanar So how can I can tell the install script to not forcefully reinstall all the dependencies from source?

developandplay, 2022-06-25:
The alternative solution would be a way to specify the wheel sources for pandas, numpy and pynq in the finn-examples installer so that if it forcefully reinstalls everything it at least happens quickly.

developandplay, 2022-06-25:
Ok so I didn't manage to tell the finn-examples installer to not reinstall everything.
But I did manage to tell it to use the precompiled wheels for numpy, pandas and cffi.
The steps I took are the following:
1. Download `numpy-1.23.0-cp38-cp38-linux_armv7l.whl`, `pandas-1.4.3-cp38-cp38-linux_armv7l.whl` and `cffi-1.15.0-cp38-cp38-linux_armv7l.whl` from [here](https://wheel-index.linuxserver.io/ubuntu/).
2. Run `pip3 install --find-links ??? git+https://github.com/Xilinx/finn-examples.git@dev -v` where `???` is the path where you downloaded the whl files to. This will make the installer use your precompiled files instead of trying to build everything.

@maltanar Maybe their is a prettier/automated way to do this...

AlmogDavid, 2022-07-13:
@developandplay  you are a life saver, thank you.

pinootto, 2022-09-03:
> Ok so I didn't manage to tell the finn-examples installer to not reinstall everything. But I did manage to tell it to use the precompiled wheels for numpy, pandas and cffi. The steps I took are the [...]

I have executed the above 2 steps, but pip3 install is still building all the packages, ignoring the downloaded whl files in the directory indicated by --find-links

here is my command:

`root@pynq:~# pip3 install --find-links /root/tmp git+https://github.com/Xilinx/finn-examples.git@dev -v`

and here is the directory with whl files:

root@pynq:~/tmp# v
total 62432
-rw-r--r-- 1 root root   360266 Oct 21  2021 cffi-1.15.0-cp38-cp38-linux_armv7l.whl
-rw-r--r-- 1 root root 19660516 Jun 22 23:58 numpy-1.23.0-cp38-cp38-linux_armv7l.whl
-rw-r--r-- 1 root root 43905267 Jun 23 16:33 pandas-1.4.3-cp38-cp38-linux_armv7l.whl

here is the execution of the command:

Collecting wheel
    Using cached wheel-0.37.1-py2.py3-none-any.whl (35 kB)
  Collecting pynq>=2.5.1
    Using cached pynq-2.7.0-cp38-cp38-linux_armv7l.whl
  Collecting tomli>=1.0.0
    Using cached tomli-2.0.1-py3-none-any.whl (12 kB)
  Collecting packaging>=20.0
    Using cached packaging-21.3-py3-none-any.whl (40 kB)
  Collecting typing-extensions
    Using cached typing_extensions-4.3.0-py3-none-any.whl (25 kB)
  Collecting cffi
    Using cached cffi-1.15.1-cp38-cp38-linux_armv7l.whl
  Collecting pandas
    **Using cached pandas-1.4.4.tar.gz (4.9 MB)**
    Installing build dependencies: started
    Installing build dependencies: still running...
    Installing build dependencies: finished with status 'done'
    Getting requirements to build wheel: started
    Getting requirements to build wheel: still running...
    Getting requirements to build wheel: finished with status 'done'
    Preparing metadata (pyproject.toml): started
    Preparing metadata (pyproject.toml): finished with status 'done'
  Collecting numpy
    Using cached numpy-1.23.2-cp38-cp38-linux_armv7l.whl
  Collecting pyparsing!=3.0.5,>=2.0.2
    Using cached pyparsing-3.0.9-py3-none-any.whl (98 kB)
  Collecting pycparser
    Using cached pycparser-2.21-py2.py3-none-any.whl (118 kB)
  Collecting pytz>=2020.1
    Using cached pytz-2022.2.1-py2.py3-none-any.whl (500 kB)
  Collecting python-dateutil>=2.8.1
    Using cached python_dateutil-2.8.2-py2.py3-none-any.whl (247 kB)
  Collecting six>=1.5
    Using cached six-1.16.0-py2.py3-none-any.whl (11 kB)
  Building wheels for collected packages: pandas
    Building wheel for pandas (pyproject.toml): started
    Building wheel for pandas (pyproject.toml): still running...
    Building wheel for pandas (pyproject.toml): still running...
    Building wheel for pandas (pyproject.toml): still running...
    Building wheel for pandas (pyproject.toml): still running...
    Building wheel for pandas (pyproject.toml): still running...

developandplay, 2022-10-11:
> 2. Run `pip3 install --find-links ??? ***@***.*** -v` where `???` is the path where you downloaded the whl files to. This will make the installer use your precompiled files instead of trying to build everything.

On a quick first glance this looks like a permission issue. I think you
might have downloaded the files as a root user instead of the normal user.

tuf22191, 2022-11-21:
So this is also happening for my Pynq Z1 board I got recently. Is there any update on this? 

I am posting below questions for other folks to read as documentation: 
Question:
1.  If I pulled in the Github package, what do we modify so that the setup.py file actually behaves the right way doing an install of just the package and not the dependencies?
2. If I have to do the workaround method, how do I find the dependency version of the current version of the package? So for numpy, pandas, cffi, what versions are needed for current version (0.5 in here: https://github.com/Xilinx/finn-examples/tags). Also are those the only packages required?

mmrahorovic, 2023-02-14:
Hi @developandplay, @tuf22191, @pinootto, @AlmogDavid, thank you for your post!

Unfortunately, I have not tested the installation on a Pynq-Z2 board. I'm not sure how many of the above questions are still relevant, but I'll try to give an answer.
You're right indeed, only the source distribution of FINN-examples is available on PyPI. So when you want to install it, you'll end up first building it, for which `pip` creates an isolated environment to build/install the dependencies. If you want to disable this option (i.e. to disable the isolation), you can pass the flag `--no-build-isolation`. Note that this would require that you have the dependencies already installed in your environment.
Please give the latest [FINN-examples set-up instructions](https://github.com/Xilinx/finn-examples#quickstart) a try and let me know if you still run into the same issues or have any other questions!

penrosed, 2025-01-16:
Closing this issue due to it being inactive. Please feel free to re-open if you have any further questions!

closed 2025-01-16 by penrosed