Xilinx/finn-examples #39: How to use bincop-cnv (mask detection) on Pynq-Z2 (version 2.7)?

pinootto, 2022-09-04:
I have installed finn-examples on Pynq-Z2 (version 2.7), but I cannot run the notebook 3_binarycop_mask_detection.ipynb

The error is:
`Exception: Bitfile for model = bincop-cnv target platform = Pynq-Z2 not found. Looked in: ['/usr/local/share/pynq-venv/lib/python3.8/site-packages/finn_examples/bitfiles/Pynq-Z2/bincop-cnv.bit', '/usr/local/share/pynq-venv/lib/python3.8/site-packages/finn_examples/bitfiles/bitfiles.zip.d/Pynq-Z2/bincop-cnv.bit']`

Is there a way to run this example on Pynq-Z2 too?

mmrahorovic, 2023-02-14:
Hi @pinootto, great question!
Since we don't provide the build script for the BinaryCoP facemask detection notebook example, you cannot rebuild this model for the Pynq-Z2 board. The bitfile that is provided is targeting the Pynq-Z1 board.

penrosed, 2025-01-16:
Closing this issue due to it being inactive. Please feel free to re-open if you have any further questions!

closed 2025-01-16 by penrosed