Xilinx/finn-examples #44: ResNet-50 Python Brevitas Model

oshears, 2022-11-16:
Hi FINN Team,

I am currently working on a project that is interested in recreating the quantized .onnx file used for the ResNet-50 FINN example here:

`https://github.com/Xilinx/finn-examples/tree/main/build/resnet50`

Is there any way that the notebook/python code for creating the quantized, brevitas version of the ResNet-50 model can be shared? 

I am interested in trying to adapt the network to fit on a ZCU102/ZCU104.

mmrahorovic, 2023-02-14:
Hi @oshears, good question and great to hear about your project!
The Brevitas code for ResNet-50 is indeed currently not released, but we are working on it and we will let you know in the upcoming months when it is ready!

MRFiruzabadi, 2023-10-09:
Hi @mmrahorovic, Did you manage to release the Brevitas code for ResNet-50?

mmrahorovic, 2023-10-09:
Hi @MRFiruzabadi
The code isn't yet released. However, in the meantime, you can find the ResNet-18 model in Brevitas, which can serve as a starting point for extending this to a ResNet-50 model: https://github.com/Xilinx/brevitas/blob/dev/src/brevitas_examples/bnn_pynq/models/resnet.py

Adib6637, 2024-05-01:
Hi, has the code for resnet 50 been released?

penrosed, 2025-01-16:
Unfortunately, the code for quantising ResNet-50 still hasn't been released.

However, if you still want to use FINN with residual networks, you should check out [this comment on issue #30](https://github.com/Xilinx/finn-examples/issues/30#issuecomment-2589692593). It outlines how you could use Brevitas' ResNet-18 example, in combination with a ResNet-18 build script, to get a decent idea of where to start.