Xilinx/finn-examples #45: Installation under PYNQ v3.0.1 for ZCU104 is broken

verderog, 2022-12-15:
FYI--

I attempted to install this under the latest PYNQ v3.0.1 image for the ZCU104.  I performed the following steps:

   - sudo python3 -m pip install --upgrade pip setuptools
   - sudo pip3 install finn-examples

The second step above failed with the error: `No module named 'pynqutils'`

The finn-examples README calls out PYNQ v2.6.1, but only v2.6.0 is readily available at http://www.pynq.io/board.html.  Re-imaging my microSD with v2.6.0 and re-running the commands above completed successfully.

Figured I'd post the issue here in case anybody else runs into this.

mmrahorovic, 2023-02-14:
Hi @verderog!

Thank you for the post. We have added [new instructions](https://github.com/Xilinx/finn-examples#zynq) on installing FINN-examples with Pynq v3.0.1. Please let me know if this resolves your issue.

verderog, 2023-02-20:
@mmrahorovic I hope to get to trying out the new instructions this week.  I'll keep you posted on my progress.  Thanks.

verderog, 2023-02-21:
I've been able to confirm that the install steps can now be completed successfully.  Unfortunately MobileNet-v1, the model I'm testing, doesn't seem to work with Pynq 3.0.1.  I see this is a known issue (#40), however, so I'm going to go ahead and mark this issue close.

closed 2023-02-21 by verderog

mmrahorovic, 2023-02-22:
Hi @verderog 
Glad to hear that the installation works!
That's indeed right unfortunately. If you want to generate a bitfile for MobileNet-v1 for the ZCU104, you could try out the 'FINN-driver' & build flow from FINN-examples v0.0.5 (with Pynq v2.6.1). That one _should_ work, although we haven't tested it lately.