Xilinx/finn-examples #65: How to target a different board/FPGA than the examples?

davidfhorowitz, 2023-09-06:
In the documentation it says you can re-target the examples for any FPGA with enough room. I would like to target a ZCU111 RFSoC so I can insert the IP into a custom design. I believe I need to alter the "build.py" file but I'm not clear on which lines need to be altered and how to specify ZCU111 rather than the default ZCU104. Can you provide guidance? 

Thank you

davidfhorowitz, 2023-09-07:
Is it enough that I just switch the part in the Vivado GUI and regenerate everything?

fpjentzsch, 2023-10-12:
Hi, you should be able to change the part in Vivado after the fact, at least if the device is from the same family and the HLS-generated code is compatible.

To see which Boards you can select in the build.py, see https://github.com/Xilinx/finn/blob/dev/src/finn/util/basic.py

For an example of how to add support for new boards, you could refer to this PR: https://github.com/Xilinx/finn/pull/787

closed 2024-08-16 by auphelia