Xilinx/finn-examples #70: Vivado project files for examples

siwave, 2023-12-09:
Are the Vivado projects for the pre-built examples available for download?

Thanks

mmrahorovic, 2024-01-08:
Hi @siwave,

The Vivado projects aren't available for download. However, for each pre-built network example, we have provided steps to rebuild the bitfile, which can be found [here](https://github.com/Xilinx/finn-examples/commits/dev). The Vivado project of the stitched-IP can then be found under `$FINN_HOST_BUILD_DIR`. Depending on which example network you're interested in and presuming you only want the Vivado project of the stitched-IP, you can add an additional flag called "stop_step" in the `DataflowBuildConfig` to run the build flow up until creating the stitched-IP. Since this depends slightly on the custom steps of the networks, I've listed which step generates the stitched-IP:

- `bnn-pynq`, `cybersecurity-mlp`, `kws`, `vgg10-radioml`: "step_create_stitched_ip".
- `mobilenet-v1`: "step_create_stitched_ip" when targeting Zynq platforms, "step_synthesize_bitfile" when targeting Alveo platforms.
- `resnet50`: "step_synthesize_bitfile".

You would need to modify the `build.py` and add, for example for the `bnn-pynq models`, the line `stop_step="create_stitched_ip"` as part of the parameters of the [DataflowBuildConfig](https://github.com/Xilinx/finn-examples/blob/b1adbdb004bfb9aacb08fa848ca0298fd115b488/build/bnn-pynq/build.py#L54-L63).

Additionally, if you want to synthesize the stitched-IP, you can set the parameter `enable_stitched_ip_synth=True` in the DataflowBuildConfig. Note however that this will increase the execution time of the build flow.

Hope this helps!

penrosed, 2025-01-16:
Closing this issue due to it being inactive. Please feel free to re-open if you have any further questions!

closed 2025-01-16 by penrosed