Xilinx/finn-examples #73: Working with Custom model

ppccww0201, 2024-01-05:
Hi.

I tried the [vgg10-radioml example](https://github.com/Xilinx/finn-examples/tree/main/build/vgg10-radioml). I synthesized the vgg10-radioml model with the command `./run-docker.sh build_custom $FINN_EXAMPLES/build/vgg10-radioml` using FINN.

Ultimately, I want to export a custom model, constructed based on Brevitas (or any other framework), as ONNX and synthesize it with FINN, rather than using the provided vgg10-radioml ONNX. I have three questions and I don't necessarily need answers to all of them. I hope to receive advice that helps me achieve my ultimate goal.

1. After exporting my custom Brevitas model to ONNX and replacing the ONNX file in `$FINN_EXAMPLES/build/vgg10-radioml`, I encountered numerous errors. Is the `$FINN_EXAMPLES/build/vgg10-radioml/build.py` specifically written for the vgg10-radioml model? Can I not synthesize my custom model using this `build.py` code?

2. If there is code available for the vgg10-radioml model before exporting it to ONNX, I think I can build my custom model based on that and synthesize it with FINN smoothly. The vgg10-radioml model code likely incorporates an understanding of FINN and adheres to many FINN constraints, making it easier to transplant my custom model. Could I obtain the code for the vgg10-radioml model? (Even if it's not based on Brevitas)

3. It seems that the `brevitas.onnx.export_finn_onnx` function is no longer supported. Is this correct? I am currently trying to use `brevitas.export.export_onnx_qcdq` and `brevitas.export.export_brevitas_onnx`. Which one should I use, or is there a more suitable export function?

Thank you.

fpjentzsch, 2024-01-08:
Hi,
I think the FINN Github discussions might be a better place to discuss this: https://github.com/Xilinx/finn/discussions

The model should be based on [this notebook](https://github.com/Xilinx/brevitas-radioml-challenge-21/blob/main/sandbox/notebooks/training_and_evaluation.ipynb) and you can refer to [this related thread](https://github.com/Xilinx/finn/discussions/420) for how to "pre-process" the exported model for FINN, although the export in the "FINN-ONNX" format has since been deprecated in favor of the QONNX export.

In general, you will most likely need to customize the build steps to fit your new model, as you can tell by the different "custom steps" that are used for the different finn-examples. For example, the VGG10 requires additional transformations to deal with the 1D convolution, while a ResNet will require additional care to streamline residual connections.

penrosed, 2025-01-16:
Closing this issue due to it being inactive. Please feel free to re-open if you have any further questions!

closed 2025-01-16 by penrosed