Xilinx/finn-examples #78: MobileNet Example(Maxpool Version) doesn't work

ppccww0201, 2024-03-28:
Hello!
I have successfully completed the following workflow:
1. Extract  Brevitas model at [Brevitas Example](https://github.com/Xilinx/brevitas/tree/master/src/brevitas_examples/imagenet_classification) in ONNX format using a slightly modified version of the `test_brevitas_mobilenet` function at [finn/tests/brevitas/test_brevitas_mobilenet.py](https://github.com/Xilinx/finn/blob/b3bdff118ae076cb776af6e51ddc28eeaa0d6390/tests/brevitas/test_brevitas_mobilenet.py#L58).
2. Compile the extracted ONNX Graph to HW following the example at [build/mobilenet-v1](https://github.com/Xilinx/finn-examples/tree/main/build/mobilenet-v1).

In this process, I wanted to make slight modifications to the Brevitas model at [Brevitas Example](https://github.com/Xilinx/brevitas/tree/master/src/brevitas_examples/imagenet_classification) by replacing TruncAvgPool2d with QuantMaxPool2d.
Then, I attempted to extract the modified model in ONNX format and synthesize it to HW following the example at [finn-examples/build/mobilenet-v1](https://github.com/Xilinx/finn-examples/tree/main/build/mobilenet-v1).

However, during the partitioning process (I'm not exactly sure what this operation entails), I encountered an `AssertionError: cycle-free graph violated: partition depends on itself` error(`self.partitioning(node) != partition_id` is False, so asserted).
I'm wondering meaning of the error and how I can successfully synthesize the modified model to HW. Please help!

Let me show you some details.

Modified Brevitas Model(only the modified parts are provided compared to [Brevitas Example](https://github.com/Xilinx/brevitas/tree/master/src/brevitas_examples/imagenet_classification)):
```
class MobileNet_ori(nn.Module):

    def __init__(
            self,
            channels,
            first_stage_stride,
            act_bit_width,
            weight_bit_width,
            round_average_pool=True,
            weight_quant=CommonIntWeightPerChannelQuant,
            first_layer_bit_width=8,
            first_layer_weight_quant=CommonIntWeightPerChannelQuant,
            last_layer_weight_quant=CommonIntWeightPerTensorQuant,
            last_layer_bit_width=8,
            avg_pool_kernel_size=7,
            first_layer_stride=2,
            in_channels=3,
            num_classes=1000):
        super(MobileNet_ori, self).__init__()
        init_block_channels = channels[0][0]

        self.features = Sequential()
        init_block = ConvBlock(
            in_channels=in_channels,
            out_channels=init_block_channels,
            kernel_size=3,
            stride=first_layer_stride,
            weight_bit_width=first_layer_bit_width,
            weight_quant=first_layer_weight_quant,
            act_bit_width=act_bit_width,
            activation_scaling_per_channel=True)
        self.features.add_module('init_block', init_block)
        in_channels = init_block_channels
        for i, channels_per_stage in enumerate(channels[1:]):
            stage = Sequential()
            pw_activation_scaling_per_channel = i < len(channels[1:]) - 1
            for j, out_channels in enumerate(channels_per_stage):
                stride = 2 if (j == 0) and ((i != 0) or first_stage_stride) else 1
                mod = DwsConvBlock(
                    in_channels=in_channels,
                    out_channels=out_channels,
                    stride=stride,
                    act_bit_width=act_bit_width,
                    weight_bit_width=weight_bit_width,
                    weight_quant=weight_quant,
                    pw_activation_scaling_per_channel=pw_activation_scaling_per_channel)
                stage.add_module('unit{}'.format(j + 1), mod)
                in_channels = out_channels
            self.features.add_module('stage{}'.format(i + 1), stage)
        # Exporting to torch or ONNX qcdq requires round
        avgpool_float_to_int_impl_type = 'ROUND' if round_average_pool else 'FLOOR'
        self.final_pool = QuantMaxPool2d(
            kernel_size=avg_pool_kernel_size,
            stride=1)            
        # self.final_pool = TruncAvgPool2d(
[... 119 lines cut ...]
img_torch = torch.from_numpy(img_np).float()
model = test_brevitas_mobilenet_custom_dirty(quant_mobilenet_v1_ori(), img_torch = img_torch, img_size = (3,224,224), model_name = "mobilenet_v1_max")
```

The error message:
```
Running step: step_mobilenet_streamline [1/14]
Running step: step_mobilenet_lower_convs [2/14]
Running step: step_mobilenet_convert_to_hls_layers_separate_th [3/14]
Running step: step_create_dataflow_partition [4/14]
Traceback (most recent call last):
  File "/home/vboxuser/Desktop/FINN/finn-examples-main/build/finn/src/finn/builder/build_dataflow.py", line 177, in build_dataflow_cfg
    model = transform_step(model, cfg)
  File "/home/vboxuser/Desktop/FINN/finn-examples-main/build/finn/src/finn/builder/build_dataflow_steps.py", line 379, in step_create_dataflow_partition
    parent_model = model.transform(
  File "/home/vboxuser/Desktop/FINN/finn-examples-main/build/finn/deps/qonnx/src/qonnx/core/modelwrapper.py", line 146, in transform
    (transformed_model, model_was_changed) = transformation.apply(transformed_model)
  File "/home/vboxuser/Desktop/FINN/finn-examples-main/build/finn/src/finn/transformation/fpgadataflow/create_dataflow_partition.py", line 80, in apply
    parent_model = model.transform(
  File "/home/vboxuser/Desktop/FINN/finn-examples-main/build/finn/deps/qonnx/src/qonnx/core/modelwrapper.py", line 146, in transform
    (transformed_model, model_was_changed) = transformation.apply(transformed_model)
  File "/home/vboxuser/Desktop/FINN/finn-examples-main/build/finn/deps/qonnx/src/qonnx/transformation/create_generic_partitions.py", line 120, in apply
    assert (
AssertionError: cycle-free graph violated: partition depends on itself
```

fpjentzsch, 2024-03-31:
Hi,

this error usually indicates that your graph contains incompatible nodes that FINN could not convert to HLS backend nodes. If any such node remains in the graph, dataflow partitioning fails.

Can you share an image of your onnx graph after the "convert_to_hls" step(s)?

ppccww0201, 2024-04-01:
Here is my graph image. The onnx file name is `step_mobilenet_convert_to_hls_layers_separate_th.onnx`.

[image]

ppccww0201, 2024-04-02:
I add [AbsorbConsecutiveTransposes](https://github.com/Xilinx/finn/blob/766e9fb47f0c31e82c45cfb9384d1ced8b1ddf7a/src/finn/transformation/streamline/absorb.py#L453) and [RemoveCNVtoFCFlatten](https://github.com/Xilinx/finn/blob/dev/src/finn/transformation/move_reshape.py#L9), then it works!!

Thank you so much!

ppccww0201, 2024-04-03:
> I see 3 problems:
> 
> 1. The first conv layer (Im2Col->MatMul) is not converted to HLS, most likely because its input doesn't have a quantzied datatype annotation and FINN doesn't support floating point. Did you already remove parts of the input quantization from this graph? If the input is already quantized (e.g. UINT8 for RGB image date), you may need to annotate this manually. Also, the first Mul will need to be removed or streamlined into a MultiThreshold, otherwise it will result in a floating point datatype at its output.
> 2. The Transpose->Transpose in the middle should cancel each other out, so you can streamline them with [AbsorbConsecutiveTransposes](https://github.com/Xilinx/finn/blob/766e9fb47f0c31e82c45cfb9384d1ced8b1ddf7a/src/finn/transformation/streamline/absorb.py#L453)
> 3. The Transpose->Reshape near the end looks like the transition between the conv part and the dense classification layer. You should be able to streamline it with [RemoveCNVtoFCFlatten](https://github.com/Xilinx/finn/blob/dev/src/finn/transformation/move_reshape.py#L9)

About first problem as you said, first conv layer is not converted to HLS and it was't synthesized as HW.

I use `test_brevitas_mobilenet_custom_dirty` (which I showed) to make onnx graph for my network, and I thought that
```
preproc_model.set_tensor_datatype(preproc_model.graph.input[0].name, DataType["UINT8"])
```
this code annotate that the input is UNIT8. But it hasn't done...
How can I annotate manually that the input is UINT8?

ppccww0201, 2024-04-03:
My network doesn't contain below layer, and this make the input keep floating-point.
```
            init_block = ConvBlock(
                in_channels=in_channels,
                out_channels=init_block_channels,
                kernel_size=3,
                stride=first_layer_stride,
                weight_bit_width=first_layer_bit_width,
                weight_quant=first_layer_weight_quant,
                act_bit_width=act_bit_width,
                activation_scaling_per_channel=True)
```
I think `activation_scaling_per_channel=True` option makes the input quantized..
Anyway problem solved! Thank you again.

ppccww0201, 2024-04-04:
Hi.

I work with 2 models `MobileNetV1v4_noinitConv`, `MobileNetV1v4_initConv`.
Only difference of these is that `MobileNetV1v4_initConv` has init Conv layer but `MobileNetV1v4_noinitConv` has not.
```
class CommonIntWeightPerTensorQuant(Int8WeightPerTensorFloat):
    """
    Common per-tensor weight quantizer with bit-width set to None so that it's forced to be
    specified by each layer.
    """
    scaling_min_val = 2e-16
    bit_width = None

class CommonIntWeightPerChannelQuant(CommonIntWeightPerTensorQuant):
    """
    Common per-channel weight quantizer with bit-width set to None so that it's forced to be
    specified by each layer.
    """
    scaling_per_output_channel = True

class CommonIntActQuant(Int8ActPerTensorFloat):
    """
    Common signed act quantizer with bit-width set to None so that it's forced to be specified by
    each layer.
    """
    scaling_min_val = 2e-16
    bit_width = None
    restrict_scaling_type = RestrictValueType.LOG_FP

class CommonUintActQuant(Uint8ActPerTensorFloat):
    """
    Common unsigned act quantizer with bit-width set to None so that it's forced to be specified by
    each layer.
    """
    scaling_min_val = 2e-16
    bit_width = None
    restrict_scaling_type = RestrictValueType.LOG_FP

class DwsConvBlock(nn.Module):

    def __init__(
            self,
            in_channels,
            out_channels,
            stride,
            act_bit_width,
            weight_bit_width,
            weight_quant=CommonIntWeightPerChannelQuant,
            pw_activation_scaling_per_channel=False):
        super(DwsConvBlock, self).__init__()
        self.dw_conv = ConvBlock(
            in_channels=in_channels,
            out_channels=in_channels,
            groups=in_channels,
            kernel_size=3,
            padding=1,
            stride=stride,
            weight_bit_width=weight_bit_width,
            weight_quant=weight_quant,
            act_bit_width=act_bit_width)
        self.pw_conv = ConvBlock(
            in_channels=in_channels,
            out_channels=out_channels,
            kernel_size=1,
            padding=0,
            weight_bit_width=weight_bit_width,
            weight_quant=weight_quant,
            act_bit_width=act_bit_width,
            activation_scaling_per_channel=pw_activation_scaling_per_channel)

[... 94 lines cut ...]
        self.output = QuantLinear(256, 8, bias=True, bias_quant=IntBias, weight_quant=last_layer_weight_quant, weight_bit_width=4)

    def forward(self, x):
        x = self.init_block(x)
        x = self.DWConv1(x)
        x = self.DWConv2(x)
        x = self.DWConv3(x)
        x = self.final_pool(x)
        x = x.view(x.size(0), -1)
        out = self.output(x)
        return out
```

And onnx graph after "convert_to_hls" step of `MobileNetV1v4_noinitConv`, `MobileNetV1v4_initConv`.
left: `MobileNetV1v4_noinitConv` right: `MobileNetV1v4_initConv`
[image]

As you can see, first layer of `MobileNetV1v4_noinitConv` is not converted to hls.
But `MobileNetV1v4_initConv` is successfully converted to hls, and synthesized to HW very well.
And I export these model with same function `test_brevitas_mobilenet_custom_dirty` shown in my first question block.

I think that there is no meaningful difference. Just first conv layer is depth-wise conv or not.
Isn't this trivial for converting to hls?

Thank you.